sccn/eeglab #11: Warning: Adding folders named 'resources' to the path is not supported

hamishib, 2019-04-03:
Hi, after installing Matlab r2019a, am getting a path error when eeglab starts:

`
Warning: Adding folders named 'resources' to the
path is not supported:
C:\Users\hain\GIT\various\eeglab\functions\resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 304)
  In eeglab (line 1127) 
`

iras, 2019-05-01:
Same here:

```>> eeglab
Warning: Adding folders named 'resources' to the path is not supported: /Applications/eeglab14_1_2b/functions/resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 293) 
eeglab: options file is ~/eeg_options.m
EEGLAB: adding "dipfit" v2.3 (see >> help eegplugin_dipfit)
EEGLAB: adding "firfilt" v1.6.2 (see >> help eegplugin_firfilt)
Warning: Adding folders named 'resources' to the path is not supported: /Applications/eeglab14_1_2b/functions/resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 293)
  In eeglab (line 1090) 
You are using the latest version of EEGLAB.```

iras, 2019-05-01:
Hi again, actually EEGLAB does not seem to work because of the resources folder warning. When running a script using pop_loadbv I get an error. The version 14_1_2b is the default download from the EEGLAB website but I can see from the repo that a branch version 15 is available, has that been released? Many thanks in advance.

```Warning: Adding folders named 'resources' to the path is not supported: /Applications/eeglab14_1_2b/functions/resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 293)
  In start_eeglab_reref_and_filter_EEG_dataset
  In init (line 39)
  In main (line 34) 
eeglab: options file is ~/eeg_options.m
EEGLAB: adding "dipfit" v2.3 (see >> help eegplugin_dipfit)
EEGLAB: adding "firfilt" v1.6.2 (see >> help eegplugin_firfilt)
Warning: Adding folders named 'resources' to the path is not supported: /Applications/eeglab14_1_2b/functions/resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 293)
  In eeglab (line 1090)
  In start_eeglab_reref_and_filter_EEG_dataset
  In init (line 39)
  In main (line 34) 
Undefined function or variable 'pop_loadbv'.```

nucleuscub, 2019-05-01:
@hamishib and @iras, can you please confirm this is happening to both of you with MATLAB r2109a?
Thks

nucleuscub, 2019-05-01:
OK. This has been confirmed. However, it is only a warning that should not cause any other issue. The files on this folder are not functions that EEGLAB use, but other resources and files like channel locations. @iras, the issue you are experiencing is because MATLAB cannot locate pop_loadbv.m in the path. Probably because the plugin needs to be added. 
Since this is not an error and does not lead to one either, I'll proceed to close the issue.

closed 2019-05-01 by nucleuscub

iras, 2019-05-01:
Agreed, the issues are not correlated. Thank you for your prompt reply. And, no, it is not the path, I had checked that. I will open a new issue for this one.