sccn/eeglab #14: Plugin Update fails (new plugin manager)

jiversen, 2019-04-22:
Hi Arno,

I am getting around to trying the new plugin manager--it's really nice! 

When attempting to update a plugin, I get an error in **plugin_menu line 172** that function **plugin_deactivate** is undefined. I'm using the most recent eeglab version from git.

FYI. I see that this function was deleted in commit#
```
commit 10279decccae1bd387f180e7348fde4fd61e9e40
Author: arno <arnodelorme@gmail.com>
Date:   Wed Apr 3 09:24:02 2019 -0700

    More changes before change in license

:100644 000000 98a69f62d... 000000000... D	functions/adminfunc/plugin_convert.m
:100644 000000 e8805416c... 000000000... D	functions/adminfunc/plugin_deactivate.m
:100644 000000 d4e1cf36c... 000000000... D	functions/adminfunc/plugin_managedeactivated.m
:100644 000000 d7dc3ac88... 000000000... D	functions/adminfunc/plugin_reactivate.m
```

Best,

John

nucleuscub, 2019-05-01:
Hi @jiversen, I'm guessing this issue was already addressed, right? If so, let's move to close it.
Thnks!

nucleuscub, 2019-06-04:
This problem has been fixed.
 Thanks,
R

closed 2019-06-04 by nucleuscub