sccn/eeglab #16: Undefined function or variable 'pop_loadbv'

iras, 2019-05-01:
Hi. I have installed the version of EEGLAB pointed at by the EEGLAB website (14 1 2b) in macOS Mojave (10.14.4) and running on MATLAB 2019a update 1. I downloaded it twice.

I got the error highlighted in the title.

I manually checked for the folder plugins/bva-io1.5.13 or similar but could not find it. That was previously the folder under which I could find the script pop_loadbv.m

Would it be possible to check the zip file to check that the folder has not been left behind by mistake provided that it was meant to be part of this version of EEGLAB?

Many thanks.

nucleuscub, 2019-05-02:
@iras , pop_loadbv belongs to the EEGLAB plugin bva-io. This means that the functionality is not distributed with EEGLAB (since it is developed by a third-party). To add plugins to your EEGLAB installation you can either check in File -> Manage EEGLAB extension from the EEGLAB GUI, or add the plugin folder under ../eeglab/plugins.
Hope this helps,
Ramon

closed 2019-05-02 by nucleuscub

iras, 2019-05-02:
Yes it helps, many thanks @nucleuscub 
That makes sense. I also found the plugin here: https://sccn.ucsd.edu/wiki/EEGLAB_Extensions.
Kind regards.