sccn/eeglab #17: can't read .mff EEG with no events (resting EEG)

sashakey, 2019-05-22:
### Description

EEGlab cannot import resting EEG (no events) files in .mff format
Error message: 
"EEGLAB error in function pop_mffimport() at line 88:
Not enough input arguments."

<br>
#### Steps to Reproduce

1. Open EEGLab
2. File - Import Data - Using EEGLab functions and plugins - Import Philips .mff file
3. Navigate to the location of the Philips/EGI .mff recording and open it.

<br>
#### Expected behavior:
Import the data file.

<br>
#### Actual behavior:
The importing process starts, correctly detects file duration, montage configuration, and then stops after detecting no events in the data file.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Mac OS 10.14 |
| Matlab version | 2019a |
| EEGLAB version | 2019 | - same thing happened in 15

satyagopal, 2019-07-16:
Hello all,

I am facing the same issue. Any progress yet? I would be thankful for any leads in this :)

[image]

Satya

sashakey, 2019-07-16:
I have not heard back on this issue yet.
Our current workaround is manually adding a DIN event to each file.

-Sasha

[error message in eeglab]<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F52961020%2F61290826-6af2e580-a7ea-11e9-983a-9bc7a644c728.png&data=02%7C01%7Csasha.key%40vanderbilt.edu%7Cd7d206da75f1443f627408d709e065cb%7Cba5a7f39e3be4ab3b45067fa80faecad%7C0%7C0%7C636988731693876369&sdata=b8GGiPjlQGLER9OmcI9K5XO3A4%2B430T3Bo1Xb%2BFaB4U%3D&reserved=0>

satyagopal, 2019-07-17:
Hello!

Many thanks for the quick reply! I shall try it! :)

Satya

braindrain123, 2019-07-31:
Hi All,

I'm experiencing the same issue, and it looks like there are no new updates. It appears that a workaround solution is adding a DIN event to the file...how is this done in Netstation? Many thanks in advance. I suppose I could always convert to a .mat file and import the good 'ole fashioned way, but I would like to utilize the .mff plugin.

Thanks,

J

satyagopal, 2019-08-03:
Hi there,

You can open the file in NetStation Review. Create a new set of events "EEGStart" and "EEGStop" or anything to your liking. Place the cursor at the start and end of your data and click on these markers from the Events pane. That's how I do it.

Hope it helps.

Satya

braindrain123, 2019-08-16:
Satya,

Thank you! This worked flawlessly. Appreciate the extra direction from your end.

Thanks,

J

nucleuscub, 2019-08-20:
This issue has been fixed in the new release of the plug-in MFFMatlabIO version 2.02.
Thks,
R

closed 2019-08-20 by nucleuscub

satyagopal, 2019-08-21:
Hi R,

I can only find 2.01 version of the plug-in on the sccn website here: https://sccn.ucsd.edu/wiki/Plugin_list_all. Is there any other link where the plug-in is available? 

Thanks!

Satya

dungscout96, 2019-08-21:
Hi Satya,

The new version can be found at: http://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php.

The wiki page has been deprecated. Please refer to the link above from now on for up-to-date list of EEGLAB plug-ins. 

Best,
Dung

satyagopal, 2019-08-21:
Hi Dung,

I followed this link and tried downloading the updated plug-in. But it shows this error: _The requested URL /eeglab/plugins//eeglab/plugin_uploader/uploaded_files/mffmatlabio2.02.zip was not found on this server._

Thank you for quick response!

Satya

dungscout96, 2019-08-21:
Hi Satya,

Sorry for the error. The page is updated with the new link. For your convenience, it's: http://sccn.ucsd.edu/eeglab/plugins/mffmatlabio2.02.zip

Let me know if you have any issue.

Dung

nucleuscub, 2019-08-21:
Thanks,  @dungscout96. We are closing this issue.