sccn/eeglab #21: pop_rejepoch.m: incorrectly calls pop_select if all trials are rejected

ascherme, 2019-06-13:
Using EEGLAB 14.1.2b, to reject specific trials, if all trials are marked for rejection, only 1 trial is removed. Searching on this issue brought up this older issue, which I think is similar: https://sccn.ucsd.edu/bugzilla/show_bug.cgi?id=832. I get the same results now using EEGLAB v2019.0 with MATLAB R2018a as described there when I run the code in Comment 5 on the epoched tutorial data:
EEG = pop_eegthresh(EEG,1,[1:32],-1,1,-1,1.9922,0,1);
It results in all 80 trials being marked for rejection, but only 1 trial being removed. 

Thanks!

arnodelorme, 2019-11-28:
I have fixed the problem.

closed 2019-11-28 by arnodelorme

ascherme, 2019-11-28:
Thank you very much.

Best,
Alice