sccn/eeglab #22: Extract epochs function ignores the first epoch

Velu44, 2019-06-23:
### Description

[*I want to process artifacts in a 1 second window and therefore, I create temporary epochs using eeg_regepoch function:  
EEG = eeg_regepochs(EEG,'eventtype', 'temp', 'extractepochs', 'on', 'recurrence', 1, 'limits', [0 1]); 
I find that, by doing so, I miss the first epoch (500 latency points as my Fs = 500 Hz) in the outcome. 

Screenshots: https://drive.google.com/drive/folders/1poMQwW_M-uhVyeDmxLNanPEoXfs51Ee7?usp=sharing

Related bug report in the past: https://sccn.ucsd.edu/pipermail/eeglablist/2017/012607.html

*]

<br>
#### Steps to Reproduce

1. Load the continuous eeg data
2. Plot the data and leave the figure open (to later compare)
3. Run the function in the command line 
EEG = eeg_regepochs(EEG,'eventtype', 'temp', 'extractepochs', 'on', 'recurrence', 1, 'limits', [0 1]); 
4. Plot the data again and compare this figure with (2).

5. Instead of (3), if I run this function,
EEG = eeg_regepochs(EEG,'eventtype', 'temp', 'extractepochs', 'on', 'recurrence', 1, 'limits', [0.002 1]); 
I get the expected behavior.
Note: 0.002 indicates the latency of my first event in the data.

<br>
#### Expected behavior:
[*I expect, if I have, let's say [0-40] seconds of data, I get 40 epochs of 1 second each.*]

<br>
#### Actual behavior:
[*Instead, I get 39 epochs*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*R2018b*] |
| EEGLAB version | [*EEG development head*] |

nucleuscub, 2019-08-09:
Hi @Velu44 ,
Can you please run the code below and let me know what do you get. The three values displayed at the end of the script must be the same and equal to the initial number of 1s segments (40 here).
Thanks,
R

`% Define number of trials
ntrials = 40;

% Create dummy EEG set
EEG        = eeg_emptyset;
EEG.srate  = 500;
EEG.data   = rand(1,ntrials*EEG.srate + 1,1);
EEG.epoch  = 0;
EEG.trials = 1;
EEG.nbchan = 1;
EEG.pnts   = ntrials*EEG.srate + 1;
EEG        = eeg_checkset(EEG);

% Here we will generate 40 trials of 1 sec duration (500 points)
EEG1 = eeg_regepochs(EEG,'eventtype', 'temp', 'extractepochs', 'on', 'recurrence', 1, 'limits', [0 1]);

% Here we will generate 40 trials of 0.3s duration (150 points). Notice here below that 0.5 and 0.8s is relative 
% to the dummy event(temp) so trials will not contain the dummy event
EEG2 = eeg_regepochs(EEG,'eventtype', 'temp', 'extractepochs', 'on', 'recurrence', 1, 'limits', [0.5 0.8]);

% Disp
disp(['Original trials:' num2str(ntrials)]);
disp(['Case 1 trials:' num2str(EEG1.trials)]);
disp(['Case 2 trials:' num2str(EEG2.trials)]);`

arnodelorme, 2019-11-28:
I have fixed the function eeg_regepochs.m so it returns the right number of epochs. The last epoch was missing though, not the first one.

closed 2019-11-28 by arnodelorme