sccn/eeglab #23: can't manage eeglab extensions

dmedine, 2019-06-26:
### Description
Windows 10 fully updated as of June 2019, Matlab  8.6 (R2015b)
Error message:

EEGLAB error in function pop_menu() at line 39:

Struct contents reference from a non-struct array

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
https://drive.google.com/open?id=1JJNMErJe02evRCxdsUgBIyEdNU56ygyT
<br>
#### Steps to Reproduce

1. [*First Step*]
Download latest version of EEGLAB (2019_0)
2. [*Second Step*]
Open Matlab
3.
CD to folder with extracted EEGLAB
4.
type eeglab
5. 
click on File>Manage EEGLAB extensions

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

nucleuscub, 2019-06-26:
Hi @dmedine!
Yep, we are aware of this issue. It is not an EEGLAB issue though. It must be fixed with the update of the JAVA version used by MATLAB. See this bug report [here](https://github.com/sccn/eeglab/issues/20) for more details.
Hope this helps,
R

closed 2019-06-26 by nucleuscub

CitizenOneX, 2019-07-24:
Just a few more details here about the issue, since I just worked it out and the comments are closed on issue #20. Some folks might need more specific details about which JAVA versions can be used.

1. https://sccn.ucsd.edu/wiki/Plugin_list_all is an https URL, and the root certificate used by SCCN is provided by Let's Encrypt, whose root certificates are only a few years old so they aren't present in root certificate stores of old programs (e.g. older versions of JAVA.)

2. According to https://letsencrypt.org/docs/certificate-compatibility/ JAVA versions after 7u111 and 8u101 should have the certificate, and hence will allow connections to Let's Encrypt-certified https websites without complaining.

3. Old MATLAB versions ship with old versions of JAVA. You can tell which version in Matlab by typing "version -java" in the Matlab command window.

4. As you might expect from the explanation above, you can fix the issue either by: 
(a) adding the Let's Encrypt certificates to your existing JAVA keystore (e.g. https://stackoverflow.com/questions/34110426/does-java-support-lets-encrypt-certificates)
(b) pointing your Matlab to use a newer version of JAVA than the one that comes bundled with it (set MATLAB_HOME environment variable and restart, e.g. https://www.mathworks.com/matlabcentral/answers/130359-how-do-i-change-the-java-virtual-machine-jvm-that-matlab-is-using-on-windows )
(c) use a newer version of Matlab that uses a new enough version of JAVA to have the Let's Encrypt certificates 

I had a few JAVA JREs installed already on my machine and pointing my R2017a at a JRE 1.8_211 did the trick for me.

Cheers