sccn/eeglab #25: Error while managing plugins

RodrigoHache, 2019-07-06:
### Description
I was not able to load .mat files, I tried to manage the extensions in order to get more file extensions to load. I'm using the latest version, just cloned from git, and also I've just  included the sub-modules
<br>
#### Steps to Reproduce

1. Start EEGLab
2. Manage EEGLab extensions 
3. The error message pop-up: EEGLAB error in function plugin_menu()at line 41: attempts to reference field of non-structure arrays
<br>
#### Expected behavior:

<br>
#### Actual behavior:
Error pop-up window appeard 
<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Linux 5.1.15-arch1-1-ARCH |
| Matlab version | 2014b |
| EEGLAB version | |

hosseinaghaie, 2019-07-15:
[image]

hosseinaghaie, 2019-07-15:
close your matlab
download 'jre-8u211-windows-x64.exe' ,and installed it
delete all file and folder in 'C:\Program Files\MATLAB\R2014b\sys\javal\jre\win64\jre'
copy all file of 'C:\Program Files\Java\jre1.8.0_211' in to 'C:\Program Files\MATLAB\R2014b\sys\javal\jre\win64\jre'

p.s: i install matlab and jre-8u211-windows-x64.exe in default folder,

[image]

nucleuscub, 2019-07-16:
Thanks for addressing this issue, @hosseinaghaie. This is though, a duplicate of a currently closed issue. The issue will be flagged accordingly.
 Thanks,
R

nucleuscub, 2019-07-16:
See  [issue 20](https://github.com/sccn/eeglab/issues/20) for more information.

closed 2019-07-16 by nucleuscub