sccn/eeglab #26: Can't import bdf files in EEGLAB

jgg888, 2019-07-10:
To whoever can help, 

I'm having issues importing BDF files using biosig in EEGLAB and I've never had this problem before. I have the latest versions of both Matlab and EEGLAB. Here are the steps that I am following and the errors I'm getting:

File > Import data > Using the BIOSIG interface > [select file]. Then a window asking to specify the channel list, start, and duration appears:
[image]

I've never seen this window before when importing data. 1) why is this new window appearing now, and what am I supposed to input? 2) No matter what I seem to input, I get the following error:
[image]

Can someone please help me figure this out? It's probably really simple and I apologize in advance for my ignorance.

nucleuscub, 2019-08-09:
Hi @jgg888,
 Please provide EEGLAB and BIOSIG version here.
 Also, check the output of the following code:
>> which sopen
You should get something like this below.
../eeglab/plugins/biosig4octmat-3.5.0/biosig/t200_FileAccess/sopen.m
Thanks,
R

arnodelorme, 2019-11-28:
Yes, apparently BIOSIG is not installed. EEGLAB should ask you to install it though. If you figure out what the problem is and why EEGLAB does not ask you to install it, let us know.

closed 2019-11-28 by arnodelorme