sccn/eeglab #28: Baseline removed twice in newtimef.m?

JohnAtl, 2019-07-12:
Here’s the relevant code:

```
Line 1216-1222:
% ----------------
% remove baseline
% ----------------
if strcmpi(g.scale, 'log') && ~any(isnan(g.powbase))g.powbase = 10.^(g.powbase/10); end; 
P = newtimeftrialbaseln(P, timesout, 'baseline', g.baseline, 'basenorm', g.basenorm, 'trialbase', g.trialbase);
[P, baseln, mbase] = newtimefbaseln(P, timesout, 'baseline', g.baseline, 'basenorm', g.basenorm, ...
                                   'verbose', g.verbose, 'powbase', g.powbase, 'trialbase', g.trialbase, 'singletrials','on');
```

It looks like ```’singletrials’,’on’``` is ignored by `newtimeftrialbaseln` but the other processing will still be performed twice. I.e. the P output of the first call is fed back in as the P input to the second call.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | 10.14.5 |
| Matlab version | 2017b |
| EEGLAB version | v2019.0 |

nucleuscub, 2019-08-09:
Thank @JohnAtl,
 My take from the two calls is that we may want to subtract baseline first from the single trials and then subtract the average baseline across trials. However, with the right combination of the parameters, it can end up doing the same process twice.
 I'm bringing @arnodelorme  to the discussion here.
Thanks,
R

arnodelorme, 2019-11-28:
Yes, this is fine. One is trial-base baseline removal. The other is average-based. Both can be combined.

closed 2019-11-28 by arnodelorme