sccn/eeglab #32: arrow.m included with eeglab uses deprecated/undocumented functions and fails in MATLAB 2019a

JohnAtl, 2019-07-28:
### Description

The arrow.m included with eeglab uses an undocumented call that fails on MATLAB 2019a.

The new version of arrow [here](https://www.mathworks.com/matlabcentral/fileexchange/278-arrow) works correctly.

<br>
2019.0 (at least)

|   |   |
|----| ---- | 
| OS version     | macOS 10.14 |
| Matlab version | 2019a |
| EEGLAB version | 2019.0 |

nucleuscub, 2019-08-09:
Hi @JohnAtl ,
 Thanks for bringing attention to this function. I can see this function is not being used in any EEGLAB function. Only Fieldtrip use it, but it uses its own copy. So, I guess we will discuss in our team if it is necessary keeping the function in EEGLAB.
 Otherwise, we will include it.
 Thanks,
R

arnodelorme, 2019-11-28:
I have deprecated the function in EEGLAB.

closed 2019-11-28 by arnodelorme