sccn/eeglab #35: Can't switch from script to GUI - Error while evaluating UIControl Callback.

GiaRanieri, 2019-08-05:
### If I go back to the GUI and perform GUI operations, an error rises if I push a button.

Whenever I try to go back to the GUI, after performing script operations on the dataset, I can't push buttons in dialogue boxes, as the following error shows up:

> **Unable to use a value of type 'matlab.ui.control.UIControl' as an index.**
>  
> **Error using inputgui (line 218)**
> **Error while evaluating UIControl Callback.**

<br>
#### Example (I'll shorthand something to make it lighter)

1. [ALLEEG, EEG, CURRENTSET, ALLCOM] = eeglab; 
EEG = pop_loadbv(rootfold, sprintf('MMM_000%i.vhdr',set));
pop_newset(ALLEEG, EEG, 0,'setname', sprintf('s0%i_raw',set), 'gui','off');
EEG = pop_resample( EEG, 250);
EEG = pop_chanedit(EEG, 'lookup','filelocation');
[ALLEEG, EEG, CURRENTSET] = pop_newset(ALLEEG, EEG, 1,'setname',sprintf('s0%i_1hpf',set),'gui','off'); % ICA dataset
EEG = pop_eegfiltnew(EEG, 'locutoff',1);
EEG = pop_cleanline(EEG, ... % parameters)
oriEEG = EEG;
EEG = clean_rawdata(EEG, 5, -1, 0.80, 'off', 8, 0.25);
EEG = pop_interp(EEG, oriEEG.chanlocs, 'spherical');
EEG = pop_reref( EEG, [29 30] ,'keepref','on');
EEG = pop_eegchanoperator( EEG, {'ch65=ch31-ch32 label biVEOG'}); %bipolar ocular channels
EEG = pop_eegchanoperator( EEG, {'ch66=ch57-ch58 label biHEOG'});
command = '[EEG LASTCOM] = eeg_eegrej(EEG,eegplot2event(TMPREJ, -1));';
eegplot(EEG.data, 'srate', EEG.srate,  'events', EEG.event, 'command', command, 'winlength', 100);
2. [*As the software plots the EEG data scroll, I can't push the Norm and Stack buttons on the right, as the error above appears.*]
3. [*The same happens if I run > pop_runamica(EEG), where I can't press the button to run the algorithm.*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*Matlab 2019a*] |
| EEGLAB version | [*eeglab2019_0*] |

nucleuscub, 2019-08-09:
@GiaRanieri, Which plug-ins have you in EEGLAB when this happens? Can you narrow the issue when using only natives EEGLAB functions (not plug-ins)? Otherwise, please provide a list of the plug-ins to reproduce the problem.
 Thanks,
R

GiaRanieri, 2019-08-09:
This is my plug-in list -

> Retrieving plugin versions from server...
> EEGLAB: adding "AMICA" v1.5.1 (see >> help eegplugin_amica)
> EEGLAB: adding "Cleanline" v1.04 (see >> help eegplugin_cleanline)
> EEGLAB: adding "ERPLAB" v7.0 (see >> help eegplugin_erplab)
> EEGLAB: adding "PrepPipeline" v0.55.3 (see >> help eegplugin_prepPipeline)
> EEGLAB: adding "bva-io" v1.5.13 (see >> help eegplugin_bva_io)
> EEGLAB: adding "clean_rawdata" v1.00 (see >> help eegplugin_clean_rawdata)
> EEGLAB: adding "dipfit" v3.2 (see >> help eegplugin_dipfit)
> EEGLAB: adding "firfilt" v2.3 (see >> help eegplugin_firfilt)
> You are using the latest version of EEGLAB.

The issue happens both when callling native EEGLAB functions (pop_eegplot) and plug-in functions (pop_selectcomps) with AMICA. The error shows up when interacting with buttons in the GUI.

Thanks for listening,
GR

nucleuscub, 2019-08-09:
Thanks, @GiaRanieri. Let me reformulate my previous question. Is the issue persisting after uninstalling the following plug-ins?

EGLAB: adding "AMICA" v1.5.1 (see >> help eegplugin_amica)
EEGLAB: adding "Cleanline" v1.04 (see >> help eegplugin_cleanline)
EEGLAB: adding "ERPLAB" v7.0 (see >> help eegplugin_erplab)
EEGLAB: adding "PrepPipeline" v0.55.3 (see >> help eegplugin_prepPipeline)
EEGLAB: adding "clean_rawdata" v1.00 (see >> help eegplugin_clean_rawdata)

Thanks,
R

GiaRanieri, 2019-08-09:
Ok, well - the issue seems to persist, even after removing eeglab altogether and reinstalling it with no additional plug-ins. I get the following message when interacting with the GUI (not always). For instance, the error rises when I try to manage extensions or when I try to delete ICA components.

> Unable to use a value of type 'matlab.ui.control.UIControl' as an index.
>  
> Error using inputgui (line 218)
> Error while evaluating UIControl Callback.

nucleuscub, 2019-08-09:
It will be hard to troubleshoot this if there is no consistent way to reproduce the issue. Also, I find very peculiar that you get the error in inputgui.m while managing extensions.  While doing that, EEGLAB does not call at any time the function inputgui.m. So... I'm kind of puzzled here...
 Can you please try installing the version in this [link](https://github.com/sccn/eeglab/archive/develop.zip) and then check if still break?
 Otherwise, please get in contact with us [here](eeglab@sccn.ucsd.edu).
Thanks,
R

arnodelorme, 2019-11-28:
Maybe you have automagic installed. Would you mind to confirm. See the report below

https://github.com/sccn/eeglab/issues/36

arnodelorme, 2019-11-28:
Duplicate of #36

arnodelorme, 2019-12-11:
@nucleuscub would you mind to try to reproduce the issue with the tutorial dataset under Windows 10 and Matlab 2019a (which I think we have installed) using both AMICA and EEGPLOT. It looks like a problem due to treating object handles as numbers. Otherwise, if we cannot reproduce the problem, we should close the bug. Tx

nucleuscub, 2019-12-11:
@arnodelorme, The script provided can not be used for debugging (paths, variables, and parameters are missing). I've tried a simplified version of the script loading the EEGLAB dataset and  invoking the last 2 lines of the original script calling eegplot,  and all seems to be fine (Using R2018b under Windows 10 and  R2019a on MACOS). Using these setup I was not able to reproduce the issue. @GiaRanieri  do you have any update on this issue?
 I will try the same after installing MATLAB 2019a and update this report... but I don't think this will modify the results.

arnodelorme, 2019-12-11:
Thanks @nucleuscub closing the bug for now. @GiaRanieri should reopen it if he can provide new data.

closed 2019-12-11 by arnodelorme

venpopov, 2020-05-07:
I ran into the same problem and after some digging I realized what caused it. I defined a variable "set" to use for something. Turns out, "set" is a pre-built function, and this overwrote it. From the second line of the original example post, you can see that it also uses "set" as a variable

nucleuscub, 2020-05-07:
Thanks for the observation, @venpopov. Indeed, that can be the issue here.  Le'ts make clear that this is not a good programming practice and there is nothing we can do to prevent this other than educate our users. This link here fits perfectly for this discussion.
https://www.mathworks.com/help/matlab/matlab_prog/function-precedence-order.html