sccn/eeglab #36: wrong number of arguments error while using GUI

P-Beach, 2019-08-07:
### Description
An issue just this last week started happening. I open EEGLAB in matlab as normal simply typing 'eeglab' on the command line. Then I try to open an ongoing study using the GUI and I get the following error message:
[image]

This happens on both my laptop (Macbook Air 13", 2017 running Mojave) and my iMac (21.5inch, 2010 running Sierra). It started while using MATLAB 2017b (iMac) but also started occurring on MATLAB 2019a. I temporarily fixed the issue last week by uninstalling MATLAB on both machines and reinstalling the newest version. Then it began occurring again (oddly enough much later on the Mac Air). I am running EEGLAB v2019.0 on both machines.

Note that I run EEGLAB and all associated scripts/plugins using a dropbox account such that all scripts are synced across machines. I tried pulling the EEGLAB directory on to one of the machine's desktop and received the same error.

Weirdly as well, when I close MATLAB I see this pop up.
[image]

Any thoughts?

Thanks,
Paul

nucleuscub, 2019-08-09:
Hi Paul,
 This is hard to troubleshoot with the information you are providing. We can not rule out issues that are not relevant to EEGLAB. My advice is doing the following:

1. Remove ALL toolboxes from the path. Just leave the default MATLAB toolboxes.
2. Download EEGLAB to your computer (grab the zip from GitHub) and add it to the MATLAB path. Do not install any extra plug-in to the ones that already come with EEGLAB.
3. Launch EEGLAB and check if there is an issue.

Let us know how that goes. I suspect there must be some meddling with other toolboxes.
Thanks,
R

P-Beach, 2019-08-11:
Well that does seem to have worked, at least for opening up a previously saved study.  I'll have to try adding back plugins one at a time, or something, to see what could have been causing the issue.

Thanks for your advice.

ytoh2000, 2019-09-26:
Hi Paul,

I'm having the exact same problem, and I realized that it's from the automagic plug-in. I see that if you remove youreeglabfolder/plugins/automagic1.3.0/gui from the path this error message is avoided, but I don't exactly see how any functions in those folder would be called while opening files from EEGLAB. Were you able to find any solution to this error?

Thanks,
Yong

P-Beach, 2019-09-26:
I ended up reinstalling EEGLAB after upgrading my MATLAB across both computers I use. That ended up fixing it. I think it was an issue with the EEGLAB version I had, though. I would try just deleting everything and reinstall the whole EEGLAB suite + your plugins of choice.

- Paul
--
Paul Beach DO, PhD
PGY4, Chief Resident Physician
Department of Neurology
Emory University School of Medicine

This e-mail message (including any attachments) is for the sole use of
the intended recipient(s) and may contain confidential and privileged
information. If the reader of this message is not the intended
recipient, you are hereby notified that any dissemination, distribution
or copying of this message (including any attachments) is strictly
prohibited.

If you have received this message in error, please contact
the sender by reply e-mail message and destroy all copies of the
original message (including attachments).

ytoh2000, 2019-09-27:
Hi Paul,

Thanks for the prompt response! Yes I tried reinstalling EEGLAB and adding plugins one by one and checked for the error occurring when I try to load the data file. It was automagic plugin that caused the problem. I decided not to use automagic plug-in because of this issue.

Thanks,
Yong

arnodelorme, 2019-11-28:
I have contacted the automagic plugin authors

arnodelorme, 2019-11-28:
I have also edited the text of the plugin to add a warning message in the plugin manager

JohnSmallridge, 2019-12-02:
This problem should be resolved if the latest version of the Automagic plugin is stored in the EEGLAB plugin folder. The latest version can be downloaded from here: https://github.com/methlabUZH/automagic 
Please, if you're using eeglab2019 and want to use the Automagic plugin, then please make sure you have the current Automagic version. The screenshot is from a very old Automagic version. 

Thanks

arnodelorme, 2019-12-02:
@nucleuscub  would you mind to test and confirm?

nucleuscub, 2019-12-11:
Waiting for automagic update...

JohnSmallridge, 2019-12-15:
> Waiting for automagic update...

The updated eeglab Automagic plugin has been submitted to the eeglab team. Alternatively, this update can be downloaded from https://github.com/methlabUZH/automagic and simply added to your current eeglab plugin folder (replacing your old automagic plugin folder).

arnodelorme, 2020-01-08:
@nucleuscub would you mind to check if the problem has been fixed on Windows. It works for me on Mac, so I am closing the bug for now.

arnodelorme, 2020-01-08:
Actually the plugin in EEGLAB is not the latest version. John, did you see Ramon message when rejecting the plugin

"Dear John, Your plug-in has been denied. The plug-in fails to be added to EEGLAB plug-ins. The issue is in line 32 of addAutomagicPaths.m. The function assumes Automagic is in the path 'before' adding it to it. Sincerely, EEGLAB team"

JohnSmallridge, 2020-01-08:
We've uploaded a new plugin version to eeglab which should resolve the issue. Thank you

closed 2020-01-08 by nucleuscub