sccn/eeglab #47: EEGLAB Annotations Load as 1

SilenceofThree, 2019-08-21:
### Description

Hi

[*For some reason when I import my .edf file to EEGLAB my annotations are all named 1. The file is able to be viewed perfectly with Polyman, all of the annotations load as named. However when I open it in EEGLAB, the annotations all are named 1, which means I cannot separate the data into individual sleep cycles. If anyone knows how to solve this problem, it would be greatly appreciated.*]

Sample File: [https://drive.google.com/open?id=1O_UCN6RLUYsvV1KZK-hcYhJPb2C69YwO](url)

[image]

[*As seen in image, the red annotations are all named 1*]

<br>
#### Steps to Reproduce

1. [*Import file to EEGLAB*]
2. [*Plot Data*]

<br>
#### Expected behavior:
[*Annotations to load as named*]

<br>
#### Actual behavior:
[*Annotations load as 1*]

<br>
#### Versions

|   |   |
|----| ---- | 

| Matlab version |[* R2019a*] |

nucleuscub, 2019-08-21:
@SilenceofThree , 
 Please check the link to the data. Seems to be broken.
R

SilenceofThree, 2019-08-22:
It works perfectly if you copy the URL into a new tab, not sure why it isn' working if you click on it.

I'll try to add it here again: https://drive.google.com/open?id=1O_UCN6RLUYsvV1KZK-hcYhJPb2C69YwO

SilenceofThree, 2019-08-22:
Thanks for letting me know, it should be working now

arnodelorme, 2019-11-28:
This is now supported in EEGLAB development version (simply download the zip on Github). However, in the file you provided, there was only one sleep stage (sleep stage 1). If this is not the case, please let us know.

arnodelorme, 2019-11-28:
[SampleData.edf.zip](https://github.com/sccn/eeglab/files/3902493/SampleData.edf.zip)
Attaching file for future reference

closed 2019-11-28 by arnodelorme

happymuzi, 2019-12-02:
Hello there!

When using pop_biosig function in EEGLAB v2019.1 to import edf files, the following warning showed up: "Warning: EDFANNOT2EVT.M is currently not supported! If there are problems reading EDF files, download and install mexSLOAD.mex* from https://pub.ist.ac.at/~schloegl/src/mexbiosig/".

While the EEG data imported were correct, EEG.event.type only have one value [1]. When trying the development version, EEG.event.type also have one value only which was the value of the first event, and the same warning still popped up.

MATLAB is running R2018b and here is [the link](https://drive.google.com/file/d/1AUsmnkMinjBkMLMFkSwH1nPgChGoj2gO/view?usp=sharing) to a sample file.

Any ideas on how to solve this would be greatly appreciated.

mantour, 2020-03-21:
I have the same problem as happymuzi described.
After importing edf+ file into eeglab with pop_biosig function, only the first event type was imported.
The [sample data](https://github.com/sccn/eeglab/files/3902493/SampleData.edf.zip) attached in the previous discussion actually had 3 event types, namely
sleep stage 1
sleep stage 2
sleep stage W

Which can be recognized normally when opened with [EDFbrowser](https://www.teuniz.net/edfbrowser/)

but only events with type "sleep stage 1" were imported.

> Hello there!
>
> When using pop_biosig function in EEGLAB v2019.1 to import edf files, the following warning showed up: "Warning: EDFANNOT2EVT.M is currently not supported! If there are problems [...]