sccn/eeglab #51: Channel spectra and maps bao

self-me, 2019-09-12:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
[image]

| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |
[image]
这一步出现以下错误
[image]

arnodelorme, 2019-11-28:
Would you mind to provide the dataset you are using as input?

self-me, 2019-11-29:
Thank you. The trouble has been solved

closed 2019-12-11 by nucleuscub