sccn/eeglab #52: Warning about adding 'resources' to path

VisLab, 2019-09-20:
### Description

I just started using Matlab 2019a.  I ran eeglab from the eeglab directory and got the following two warnings:
1)
`Warning: Adding folders named 'resources' to the path is not supported:
D:\Research\FluencyProject\eeglab14_1_2b\functions\resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 293) `

2)
`Warning: Adding folders named 'resources' to the path is not supported:
D:\Research\FluencyProject\eeglab14_1_2b\functions\resources 
> In path (line 109)
  In addpath (line 86)
  In eeglab>addpathifnotinlist (line 2059)
  In eeglab>myaddpath (line 2088)
  In eeglab (line 293)
  In eeglab (line 1090) 
`

<br>
#### Actual behavior:
Just a warning ---- this must be new to R2019a

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 7] |
| Matlab version | [R2019a] |
| EEGLAB version | [14_1_2b] |

olafdimigen, 2019-11-11:
Getting the same warning using same MATLAB/EEGLAB version and Win 10

arnodelorme, 2019-11-28:
We have renamed the "ressources" folder "supportfiles" for that reason in EEGLAB 2019.0 and 2019.1

closed 2019-11-28 by arnodelorme

linaismail, 2020-05-09:
Hello I have the same issue while using EEGLAB 2019

jcmota0, 2025-03-26:
Change the name of the "resources" folder to "supportfiles".
Go to the eeglab.m file, and change all the lines that have the "resources" folder name to "supportfile". 
Reload eeglab and you won't have the error.