sccn/eeglab #56: Down-sampling antialiasing filter in pop_resample() still create high-frequency noise as in bug 1757

yhz-1995, 2019-10-02:
### Description

It seems that the anti-aliasing filter in the new pop_resample() still creates high-frequency noise in down-sampled data as reported in [bug 1757](https://sccn.ucsd.edu/bugzilla/show_bug.cgi?id=1757).
[image]
[image]

<br>
#### Steps to Reproduce

```
EEG = eeg_emptyset;
EEG.times = 1:1000;
EEG.data = sin(2 * pi * 260 * (0:0.001:0.999));
EEG.srate = 1000;
EEG.nbchan = 1;
EEG.trials = 1;
EEG.pnts = 1000;
TMP = pop_resample(EEG, 512);
z = abs(fft(TMP.data));

figure('Color','w');
plot(EEG.times,EEG.data);
hold on;
plot(TMP.times,TMP.data);
xlabel('Time point');
ylabel('Signal');
ylim([-1.5,1.5]);
legend({'Original Signal','Downsampled Signal'},'box','off','Location','best');

figure('Color','w');
plot(linspace(0, 256, 257), 20 * log10(z(1:257) * 2 / 512));
xlabel('Frequency (Hz)');
ylabel('Amplitude Spectrum of Downsampled Signal (dB)');
```

<br>
#### Expected behavior:

No high-frequency noise injection at the highest frequency edge of the resampled data.

<br>
#### Actual behavior:

There is still high-frequency noise injection at the highest frequency edge of the resampled data.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 1809 |
| Matlab version | R2019a|
| EEGLAB version | 2019_0 |

arnodelorme, 2019-10-03:
Ramon, would you mind to have a look at this?
Cheers,

Arno

arnodelorme, 2019-12-11:
@widmann what do you think of this?

widmann, 2019-12-11:
This is expected behavior. Also anti-aliasing filters show edge effects. This particular kind of transient effect is most commonly observed in notch filters (where typically high amplitude harmonic oscillations have to be removed as also here). The anti-aliasing filter needs some time to fully remove the above new Nyquist frequencies and some aliasing may therefore still be observed at the edges. The only way to avoid this is to use a different type of (mirror) padding. This however, may distort phase information across the full frequency spectrum and introduce low frequency artifacts instead. DC padding was therefore preferred over other types of padding. I suggest closing.

nucleuscub, 2019-12-11:
Thanks for your comment, @widmann.  We are closing this Issue.
R

closed 2019-12-11 by nucleuscub