sccn/eeglab #61: binica.m is buggy on Windows because of some system calls

rq-Chen, 2019-10-18:
- the important function of `binica.m` is originally inperfect in these ways:
  - It **contains a bug** in line 320 that when executing system command `!ICABINARY<pwd/scriptfile`, it assumes that `ICABINARY` and `pwd` contain no space character, which is very likely wrong. Adding several double quotes will solve the problem of `ICABINARY`, but since `binica.exe` is also buggy and uses some other paths as parameters, the only way is to **change the name of your working folder so that its path contains no space!!!**
  - Other bugs in `binica.m` are relevant to system calls in windows that are different with those LINUX, namely `ls` (for windows the alternative is `dir`) in line 303/349/351 and `rm` (for windows should be `del`) in line 366. However, the script contains other bugs with respect to the variables `scriptfile` `weightsfile` `spherefile` and `datafile`, where the paths are seperated by '/' (instead of '\\\\') and leads to erroneous parameter parsing on Windows. Fortunately, these commands are not so crucial and we can simply **comment the lines 303/349/351/366** (and remove binicaNUM.xx manually afterwards).

Of course, it will be much more convenient if further version of binica.m can take them into consideration, and become completely compatible with Windows.

Thanks a lot!

arnodelorme, 2019-11-28:
We could check for spaces in the ICABINARY variable.

arnodelorme, 2019-12-11:
Would you mind submit an updated and tested binica.m function?

arnodelorme, 2019-12-11:
I have updated binica.m eaaf3ae47 and it is now fully compatible with Windows.

Arno

closed 2019-12-11 by arnodelorme