sccn/eeglab #62: Problem of importing data

yeonamon, 2019-11-06:
### Description

There were some issues related to import data, so I deleted eeglab and reinstalled it twice.
For importing, I used this path 'file - import data - using the FILE-IO interface'.
Then the notice 'Fileio extension is not installed. Do you want to download it now?' comes up, and when I click 'Yes'... there is an error,

'EEGLAB error in function plugin_askinstall() at line 78:

Struct contents reference from a non-struct array object.'

as same as the image below.
[image]

How can I fix it? I also installed another version of MATLAB(2019), it also doesn't work. 
<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [64-bit Operating System, x64-based processor] |
| Matlab version | [R2017a] |
| EEGLAB version | [2019, latest ver.] |

arnodelorme, 2019-11-28:
Would you mind to try with EEGLAB 2019.1 instead of 2019.0?

arnodelorme, 2019-11-28:
I have checked and could reproduce the problem in 2019.0 under Windows (Mac and Unix works). The issue has to do with checking URL certificates in newer versions of Matlab and cannot be easily fixed. EEGLAB 2019.1 uses a different method so it fixes the issue.

closed 2019-11-28 by arnodelorme