sccn/eeglab #68: EEGLAB won't open - maximum recursion error

adabelsdsu, 2019-11-13:
### Running Matlab 2018a and EEGLab v 14_1_2b. It used to work fine but now won't open. The error I get is that it cannot access the local eeg_options file and maximum recursion limit of 500 reached. I've uninstalled and re-installed both Matlab and EEGLab with the same result. Thoughts?

[image]

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
[image]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2019-11-28:
Yes, it is a strange problem. My intuition is that either eeg_options.m or eeg_optionsbackup.m is corrupted. We would need to be able to set a breakpoint inside eeglab_options.m to figure out what the problem is.

You can try to go into the path and remove all non original Matlab paths.

Also under Matlab, you can type "which eeg_options.m" and delete the file.

Let us know,

Arno

arnodelorme, 2019-12-11:
And you can also update the EEGLAB 2019.1 (EEGLAB 14.2 is about 3 years old). Closing issue for now.

closed 2019-12-11 by arnodelorme