sccn/eeglab #69: EEGLAB error in function pop_writeeeg()

serdark, 2019-11-14:
Hi all,

I got this error when I tried to export my data in .edf . I tried to install the neuroscanio plugin, it didn't work.

<br>

1. Open an EEG (epoched)
2. File->export->data to EDF/etc
3. This dataset contains epochs. Do you want to export the concatenated data epochs? YES (If NO, it doesn't do anything)

<br>

It give me the error:
EEGLAB error in function pop_writeeeg()

<br>
#### Actual behavior:
[*What actually happened*]

<br>

|   |   |
|----| ---- | 
| OS version     | windows 10|
| Matlab version | [R2018b] |
| EEGLAB version |v2019.0 |

serdark, 2019-11-14:
update: I tried with a file that was already concatenated. It's the same.

jmkendallbar, 2019-11-19:
Hello, I am also having this problem. I am using the latest version of EEGLAB (v2019.0), with Matlab version 2017a, and Windows 10. 

I was trying to export the following dataset (attached as a .zip) as an .edf. My dataset has 7 channels of electrophysiological data which has already been processed using the "runica" function. However, this error occurs even when I try to export the original, unprocessed data.
[EEGLAB Dataset.zip](https://github.com/sccn/eeglab/files/3866338/EEGLAB.Dataset.zip)

My friend was able to export as an .edf by using a previous version (14.1.2). 

Thank you for any help you can provide!

jmkendallbar, 2019-11-19:
Here is the error message.
[image]

arnodelorme, 2019-11-28:
Downloading the latest version 2019.1 fixes the problem. See also this message

https://sccn.ucsd.edu/pipermail/eeglablist/2019/014463.html

https://sccn.ucsd.edu/pipermail/eeglablist/2019/015007.html

closed 2019-11-28 by arnodelorme