sccn/eeglab #70: Latest EEGLAB compilation on Mac OS Catalina 10.15.1: "Symbol not found: _mecab_get_feature"

martinfrasch, 2019-11-18:
Successfully install the latest built just released.
Get this error message when launching the GUI from the terminal:

/Applications/EEGLAB/application/run_EEGLAB.sh /Applications/MATLAB/MATLAB_Runtime/v95
------------------------------------------
Setting up environment variables
---
DYLD_LIBRARY_PATH is /Applications/MATLAB/MATLAB_Runtime/v95/runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/sys/os/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/bin/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/extern/bin/maci64
dyld: Symbol not found: _mecab_get_feature
  Referenced from: /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
  Expected in: /Applications/MATLAB/MATLAB_Runtime/v95/bin/maci64/libmecab.dylib
 in /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
/Applications/EEGLAB/application/run_EEGLAB.sh: line 27: 41030 Abort trap: 6           "/Applications/EEGLAB/application/EEGLAB.app/Contents/MacOS/EEGLAB"

--

I reinstalled the matching 9.5 runtime directly from Matlab to be sure. Installed the runtime without errors. Still the same issue. Please help troubleshoot. For the reference, I ran the previous compiled EEGLAB on my Mac with no issues. 

Thank you.

PS. run_EEGLAB.sh I call above is displayed below:

#!/bin/sh
# script for execution of deployed applications
#
# Sets up the MATLAB Runtime environment for the current $ARCH and executes 
# the specified command.
#
exe_name=$0
exe_dir=`dirname "$0"`
echo "------------------------------------------"
if [ "x$1" = "x" ]; then
  echo Usage:
  echo    $0 \<deployedMCRroot\> args
else
  echo Setting up environment variables
  MCRROOT="$1"
  echo ---
  DYLD_LIBRARY_PATH=/Applications/MATLAB/MATLAB_Runtime/v95/runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/sys/os/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/bin/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/extern/bin/maci64;
  export DYLD_LIBRARY_PATH;
  echo DYLD_LIBRARY_PATH is ${DYLD_LIBRARY_PATH};
  shift 1
  args=
  while [ $# -gt 0 ]; do
      token=$1
      args="${args} \"${token}\"" 
      shift
  done
  eval "\"${exe_dir}/EEGLAB.app/Contents/MacOS/EEGLAB\"" $args
fi
exit

arnodelorme, 2019-11-28:
It seems to be a OSx Catalina issue. We are investigating.

arnodelorme, 2019-12-11:
@nucleuscub we should try to compile with Matlab 2019b on Catalina

https://www.mathworks.com/matlabcentral/answers/468831-is-matlab-compatible-with-macos-catalina

Mersina, 2019-12-14:
Did updating to Matlab 2019b work? I have a similar issue: installed Runtime 9.5 without issue and can run the .sh files from compiler without issue using my Matlab application, but get this error when I try to run using Runtime: 

Setting up environment variables
---
DYLD_LIBRARY_PATH is .:/Applications/MATLAB/MATLAB_Runtime/v95/runtime/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/bin/maci64:/Applications/MATLAB/MATLAB_Runtime/v95/sys/os/maci64
dyld: Symbol not found: _mecab_get_feature
  Referenced from: /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
  Expected in: /Applications/MATLAB/MATLAB_Runtime/v95/bin/maci64/libmecab.dylib
 in /System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/CoreNLP
./run_test_createDataSummary.sh: line 29: 44216 Abort trap: 6

martinfrasch, 2020-01-27:
still seeing the same issue after updating to Matlab 2019b compilation.

arnodelorme, 2020-02-19:
@nucleuscub , would you mind to try recompiling with 2019b now that it is available and see if it fixes the issue?

dannyzhang86, 2021-07-13:
Did this issue solve or not? I mean the same one...

martinfrasch, 2021-07-14:
EEGLAB v2021.0 runs fine on macOS Big Sur 11.4.

closed 2021-07-14 by martinfrasch

dannyzhang86, 2021-07-16:
> EEGLAB v2021.0 runs fine on macOS Big Sur 11.4. 
It was fixed after re-compiled