sccn/eeglab #72: Eprime import

arnodelorme, 2019-11-21:
I am having difficulties uploading e-prime information to my data set, 

The .bdf file as well as the e-prime data files are in this link:
https://drive.google.com/open?id=1V13Vku5WWII_nc_iBYciUuHXcNGU_Gm_

We need to check the e-prime import works with this file.

mspape, 2019-11-21:
Hi Arnaud,
Your link 404s. But I'm also curious as to why you would want to import an "e-prime" file (the experiment file or data file?) in EEGLAB. Surely the triggers are already in the BDF file itself, or is there some sort of plugin out there that magically integrates a bunch of stuff? 
Best,
Michiel

(asking this as someone writing a book about E-Prime & EEGLAB!)

arnodelorme, 2019-11-21:
Yes, the e-prime file can have additional information that might not be in the BDF file.

nucleuscub, 2019-12-12:
@arnodelorme,  the link here is broken

arnodelorme, 2024-02-10:
Closing for now.

closed 2024-02-10 by arnodelorme