sccn/eeglab #77: clean_artifacts distance error

karhei, 2019-11-22:
### Description
EEGLAB2019:
using clean_artifacts with ASR / BurstCriterion will lead to an error in line 230 of clean_artifacts.m: 
    if ~strcmpi(distance, 'euclidian')    
which loads Matlab's distance function.
<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [2016a] |
| EEGLAB version | [2019_1] |

umarshahdin, 2019-12-07:
i'm also facing the same issue could you please tell me how to solve this issue

arnodelorme, 2019-12-07:
I have issue a new version 2.1. Let me know if it works for you (if it does not, please submit a new issue directly to the clean_rawdata repository https://github.com/sccn/clean_rawdata/issues).

closed 2019-12-07 by arnodelorme

umarshahdin, 2019-12-07:
Hi @arnodelorme,
It’s working properly now.

I want you to make a video session on ASR which is not available on your youtube channel. Could you please do this?
It’s a kind request.

Thank you
Umar.

arnodelorme, 2019-12-07:
Yes, I will try to do that before the end of the year.