sccn/eeglab #84: uigetfile2 and uiputfile2 memorize previously used directories on multiuser & multisession systems

arnodelorme, 2019-11-27:
Currently uigetfile2.m and uiputfile2.m store previously used directory in a eeglab.cfg file in a /tmp directory common to all users on a computer.

This poses problems on shared computers. Users are pointed to the previous directory of whoever ran the mfile last.

Also when running several instances of matlab on a single computer, only one directory is remembered by uiputfile2 and uigetfile2.

Attached version of both files uses setpref and getpref features of matlab to set userspecific last directory. Also defines a persistent variable that allows specific values of the last directory on different instances on the same computer.

mlock is used at the beginning of the file to prevent clearing of persistent variable from memory by eeglab.

arnodelorme, 2019-11-27:
[Archive.zip](https://github.com/sccn/eeglab/files/3898921/Archive.zip)

nucleuscub, 2019-12-11:
@arnodelorme , should we move ahead with this?
The solution implies to save an extra configuration file for the paths...not sure how do you feel bout it. Let me know, what should we do. Thks, R

arnodelorme, 2019-12-11:
Yes, we should move forward.

nucleuscub, 2019-12-11:
OK. Patch has been added [in this commit](https://github.com/sccn/eeglab/commit/ece2ef2c06b8ad58ea8f063fd8a973bed7a96933)
R

closed 2019-12-11 by nucleuscub

arnodelorme, 2019-12-12:
Tx Ramon. Did you check it was working on Osx, Linux and Windows?
Cheers,

Arno

nucleuscub, 2019-12-12:
Yep, Windows 10, MACOS and Ubuntu running MATLAB R2018b. In all seems to work fine.
R

arnodelorme, 2019-12-12:
wonderful. Thanks Ramon,

Arno