sccn/eeglab #85: newtimef.m: absolute power for wavelet and FFT

arnodelorme, 2019-11-28:
Dear eeglablist subscribers,

I have a question concerning the timef function.
As I want to calculate the absolute power of my data, I set the 'baseline'
parameter to 'NaN'.
Now, first thing that concerns me, I get totally different results using
either FFT or wavelets (freqs from 2-50Hz, FFT yielded values from -25 to
10, wavelet values from 36 to 48).
Besides this, even after reading carefully through previous eeglablist
discussions on this topic, I'm still not shure how to calculate the mean
power (in uV) from the values returned by the function.
I'd appreciate any help, thanks in advance,

Ulrich Pomper

arnodelorme, 2019-11-28:
@widmann would you mind to tell us your opinion on this?

happymuzi, 2019-12-02:
The parameter pair 'baseline' and NaN only means no baseline correction. What you are looking for, I think, is the parameter pair 'scale' and 'abs'. This should be able to also control the output value. As the difference in FFT and wavelet method, my understanding is that they ought to differ since they are two different ways of time-frequency decomposition.

Hope you find this helpful.

widmann, 2019-12-03:
Sorry, this is too unspecific. And I'm not an expert for newtimef, sorry.
Andreas

closed 2021-05-21 by cll008