sccn/eeglab #91: Bug 661 - timef.m: absolute power for wavelet and FFT

nucleuscub, 2019-12-02:
### Bug report 661 in BUGZILLA

**Arnaud Delorme 2008-06-13 01:59:23 PDT**
Dear eeglablist subscribers,
I have a question concerning the timef function.
As I want to calculate the absolute power of my data, I set the 'baseline'
parameter to 'NaN'.
Now, first thing that concerns me, I get totally different results using
either FFT or wavelets (freqs from 2-50Hz, FFT yielded values from -25 to
10, wavelet values from 36 to 48).
Besides this, even after reading carefully through previous eeglablist
discussions on this topic, I'm still not shure how to calculate the mean
power (in uV) from the values returned by the function.
I'd appreciate any help, thanks in advance,
Ulrich Pomper

**[reply] [−]Comment 1Arnaud Delorme 2009-07-02 14:24:43 PDT**
Created attachment 388 [details]
Different scaling (right=fft, left=wavelet)

**[reply] [−]Comment 2Arnaud Delorme 2009-07-02 14:28:25 PDT**
This is true when we look at the atachement. When plotting the difference from baseline, the scale is similar but when plotting the absolute value, it is very different.

Arno

**[reply] [−]Comment 3Dev Sarma 2009-10-20 15:39:58 PDT**
Still a problem as of October 2009

**[reply] [−]Comment 4Dev Sarma 2009-11-10 15:29:52 PST**
Message for external help posted on wiki web site.

**[reply] [−]Comment 5Arnaud Delorme 2011-09-25 22:45:59 PDT**
Yes, still a problem. We should email the eeglablist.

Arno

[**reply] [−]Comment 6Ramon Martinez-Cancino 2014-07-25 13:32:36 PDT**
We don't use this function anymore. Should we test this with newtimef??

**[reply] [−]Comment 7Ramon Martinez-Cancino 2015-02-20 13:29:25 PST**
Yes, test with newtimef. For wavelets use wavelet factor of 0 or 1 (same as FFT) and make sure you have the same window size for both FFT and wavelets.

Arno

nucleuscub, 2019-12-03:
[image]

downeyryanj, 2020-02-27:
Note this was also Bug 13788 in bugzilla. See following link that includes an attachment for modifying the function so that the output matches what you would expect had you simply called a PSD on the same data.

https://sccn.ucsd.edu/bugzilla/show_bug.cgi?id=13788

Download the attachment "modified_newtimef". I tried it a long time ago and it resulted in the correct scale for the baseline power for me. I have since gone back to using the standard version included wtih eeglab but that's mainly because it doesn't affect the work that I do. Specifically, although the baseline power calculation is wrong, it is consistent. But most importantly, the baseline removed ERSPs are correct regardless of the baseline power scale issue. So it's something that should be fixed but probably won't cause you any real problems.