sccn/eeglab #92: Bug 1371 - std_precomp/std_ersp and powbase

nucleuscub, 2019-12-02:
### Description

**[reply] [−]DescriptionBenedikt Ehinger 2012-11-19 01:42:03 PST**
There seem to be multiple problems with using custom power-baseline ('powbase') in the study function. The problems I could identify so far are in line 345 of std_precomp. If you add the field in std_precomp( ... 'erspparams',{...'powbase',bslPowerSpectrum)} % (bslPowerSpectrum has size [nr_comp_in_study,max_freqs])  In this line the script checks for parameters in the std_ersp. Two problems occur here:  1) As this parameter check is simply done on a single channel, std_ersp will complain that there are nr_comp X freq and not 1 X freq powerSpectra of baseline. The exact error is:  'powbase should be of size (ncomps,nfreqs)' This can be changed e.g. by using:           powBaseIndx = find(cellfun(@(x) strcmp(x,'powbase'),g.erspparams)); tmp = g.erspparams{(powBaseIndx)+1}; g.erspparams(powBaseIndx+1) = {tmp(1,:)};  2) But then the std_ersp is not saving the 'powbase' command in its output and with [ tmpX tmpt tmpf g.erspparams ] = std_ersp(...) it is overwritten and gone for good. That is why you have to add it by hand again e.g.         g.erspparams(end+1) = {'powbase'};         g.erspparams(end+1) = {tmp};    I have eeglab Version 11.0.3.1b. As far as I could see this bug still exists in the newest SVN revision of eeglab.   For more details on this problem, please contact me.  Best, Benedikt

**EEGLAB version:**  *11.x*

arnodelorme, 2019-12-11:
Original bug https://sccn.ucsd.edu/bugzilla/show_bug.cgi?id=1371

arnodelorme, 2019-12-11:
This is a complex problem that is better handled with a script. I have clarified the code regarding powbase. If there is a specific application, we will consider adding functionality to support that feature at the study level.

closed 2019-12-11 by arnodelorme