sccn/eeglab #93: EDF Annotation import issue

happymuzi, 2019-12-03:
### Description

Hello there!

When using pop_biosig function in EEGLAB v2019.1 to import edf files, the following warning showed up: "Warning: EDFANNOT2EVT.M is currently not supported! If there are problems reading EDF files, download and install mexSLOAD.mex* from https://pub.ist.ac.at/~schloegl/src/mexbiosig/".

While the EEG data imported were correct, EEG.event.type only have one value [1]. When trying the development version, EEG.event.type also have one value only which was the value of the first event, and the same warning still popped up.

MATLAB is running R2018b and here is the link to a sample file.

Any ideas on how to solve this would be greatly appreciated.

<br>
#### Steps to Reproduce

1. EEG = pop_biosig(file);
2. EEG.event(:).type

<br>
#### Expected behavior:
EEG.event(:).type should be different.

<br>
#### Actual behavior:
EEG.event(:).type were the same.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 |
| Matlab version | R2018b |
| EEGLAB version | EEGLAB v2019.1 |

arnodelorme, 2019-12-11:
With the development version on Github (which has several fixes - simply download zip on Github), importing your file, I have one event of type 'Started Recording' so event annotation are supported. I am not sure about the message from BIOSIG (which is a BIOSIG issue). Let us know if the problem persist.

closed 2019-12-11 by arnodelorme

happymuzi, 2019-12-11:
Thanks for the reply. I tried the development version back when I posted this issue. The file I shared should yield at least two different event types, but in reality, EEGLAB can only find one event type. I will try the newest development version to see if this is fixed. Judging from your reply, the issue probably is still there.

arnodelorme, 2019-12-11:
I have looked at what sopen returns and there is definitely only one event type. What software did you used to see 2 event types?

happymuzi, 2019-12-11:
The sample file is converted to the EDF file from the original recording acquired by Sirenia (Pinnacle Tech). I examined the annotation in EDFbrowser, and it matched the original file. Here is [the link](https://photos.app.goo.gl/pkNXDtNdtgNnYnEY8) to a screenshot.

arnodelorme, 2019-12-11:
Yes, there is definitely an issue as sopen assumes all events are of the first type while they are indeed of different types.

arnodelorme, 2019-12-11:
I have notified Alois Schlogl, the author of BIOSIG.

mantour, 2020-03-21:
I had the same problem. I made a small modification to edfannot2evt.m and biosig2eeglabevent.m to solve it. Hope it will also work for other people.

[fix_edf_annotation.zip](https://github.com/sccn/eeglab/files/4363305/fix_edf_annotation.zip)

mantour, 2020-04-06:
I fixed a bug in my previous solution.
[edfannot2evt.zip](https://github.com/sccn/eeglab/files/4436195/edfannot2evt.zip)

RaiyaN-98, 2021-07-13:
This has happened due to this command TYP(N, 1) = 1 in edfannot2evt.m (Marked yellow in the attached image). So the event type is marking as 1 all the time. (EEGLAB v2021.0)
[image]
Though I haven't found any generalized solution. But for my data there were 3 events - T0, T1, T2. So I made a change in the edfannot2evt.m file to find out different event_num. And then assign TYP(N, 1) = event_num. It worked for me!
[image]