sccn/eeglab #94: envtopo() 'sortvar', 'pvaf' does not take effect

nucleuscub, 2019-12-03:
###  Bug 1589 in BUGZILLA

**Makoto 2014-03-04 17:24:31 PST**
There seems several things to be fixed in this function.

Makoto

**[reply] [−]Comment 1Arnaud Delorme 2014-03-05 10:13:21 PST**
Hi Makoto,

what do you mean? Can you submit a patch?
Thanks,

Arno

**[reply] [−]Comment 2Ramon Martinez-Cancino 2014-05-19 11:00:54 PDT**
Hi Makoto, 
 Can you submit a comment or script to know what to fix?
 Thanks,
 Ramon

**[reply] [−]Comment 3Arnaud Delorme 2017-09-25 20:16:24 PDT
Makoto's feedback (from talking to me):**

- there is a problem using pvaf

- even after his patch (now attached to the bug), some problem remain

- try using pvaf - does not seem to work when using pvaf (the GUI does not plot pvaf and maybe the function does not change anything). This was fixed in Makoto's version.

- other combination of parameters might not work in Makoto's fix

**[reply] [−]Comment 4Arnaud Delorme 2017-09-25 20:16:51 PDT**
Created attachment 1030 [details]
Makoto's fix

nucleuscub, 2019-12-03:
[Envtopo_updated_v3.zip](https://github.com/sccn/eeglab/files/3918286/Envtopo_updated_v3.zip)

nucleuscub, 2019-12-03:
@arnodelorme  this patch seems to be non-specific for the issue being addressed. I would not merge this file. Let's talk about what to do here.
R

arnodelorme, 2019-12-03:
I have checked the 4 sorting options in envtopo and they are all functional. There might be an error in the formula but Makoto's report and fix is not clear enough to know what it might be.

I have reviewed the differences between the functions. There are too many. It is up to Makoto to merge his changes with the latest envtopo.m function.

closed 2019-12-03 by arnodelorme