sccn/eeglab #98: Can't compute significance with pop_newcrossf

nucleuscub, 2019-12-06:
### Description

newcrossf.m issue error when trying to compute significance. Happening both in epoched and continuos data.

Error using newcrossf (line 409)
Cannot compute significance for continuous data

Error in pop_newcrossf (line 213)
eval(com)

#### Steps to Reproduce

1. Load eeglab sample set (any of them)
2. Execute: pop_newcrossf(EEG, 1, 1, 2, [-100 500],0, 'alpha', 0.05)

For any type of data, it will throw the error above.

#### Versions

|   |   |
|----| ---- | 
| OS version     | [*10.14.3*] |
| Matlab version | [*R2018b*] |
| EEGLAB version | [*develop branch as on Dec 5, 2019*] |

nucleuscub, 2019-12-06:
Related to commit:
https://github.com/sccn/eeglab/commit/79b6a4045fd5bc63897f98cb5722cb7740f2ffd2#diff-2c81245d032d0ac2f0cdd7e05ba61e0e
 Discuss with @arnodelorme

arnodelorme, 2019-12-10:
I have fixed the problem for data epochs. Pull the latest changes.

closed 2019-12-10 by arnodelorme

YouyanLee, 2020-05-18:
I got the same problem with the latest version of EEGLAB.