sccn/eeglab #99: can't load gdf file by EEGLAB

jaeeewon, 2019-12-06:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
[image]
[image]
[image]

<br>
#### Steps to Reproduce

1. I downloaded biosig, and EEGLAB. I tried to open a gdf file by eeglab in matlab
2. There is a capture img how I tried to open my file.
3. The data I used is 'bci competition graz dataset b' which is popular. (Can't upload the gdf file here)

 

<br>
#### Expected behavior:
what I wanted is just to open my gdf data by eeglab.

<br>
#### Actual behavior:
The eeglab says
''
> In pop_biosig (line 98) 
Warning: line (423: 65152,"V/s",,"",,#obsolete#) not valid
Warning: line (428: 65312,"mHg s-1","","",#obsolete#) not valid
Warning: line (430: 65376,"r.p.m",,"rotations per minute",#obsolete#) not valid
Warning: line (431: 65408,"B", "Bel", "relative power decibel",#obsolete#) not valid
Warning: line (432: 65440,"dyne s m2 cm-5","Vascular Resistance Index","dyne seconds square meter per centimetre to the power of 5",#obsolete#) not valid
Warning: line (433: 65440,"dyne*s*m짼/cm^5","Vascular Resistance Index","dyne seconds square meter per centimetre to the power of 5",#obsolete#) not valid
Warning: line (436: 65504,"T",,"<magnitude> Tesla",#obsolete#) not valid
Reading data in GDF format...
---
> In pop_biosig>readfile (line 248)
  In pop_biosig (line 165) 
Reading data file header...
---   <-### **Saying that "% # 26" is not a recognized pragma.**

''
It succeeded in my friend's computer(with the same dataset).
I tried to open gdf data in several computers, but it didn't worked.
The way I tried to open gdf file is same with my friends, so I can't figure out what is the problem.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [window64] |
| Matlab version | [R2019b(The recent version)] |
| EEGLAB version | [ eeglab2019_1,(tried with the latest previous version too)] |

arnodelorme, 2019-12-10:
Would you mind to share a data file. We cannot reproduce without a data file.

Best wishes,

Arno

arnodelorme, 2019-12-11:
I have downloaded dataset 2b as described above and imported both Mac OSX and Windows without any issue. The issue seems to be with sread so copying @alois.schloegl@gmail.com

Data Sets 2
from Graz (description 2a; description 2b)
Data sets 2a: [ GDF files zipped (420 MB) ]
Data sets 2b: [ GDF files zipped (216 MB) ]

closed 2019-12-11 by arnodelorme