sccn/eeglab #100: std_erspplot craches when the computing common baseline is set in the std_erspparams

yurimaxatf, 2019-12-10:
### Description

std_erspplot crashes when the computing common baseline is set in the std_erspparams,
however, std_erspplot works properly if the common baseline subtraction os off. 

Erros message:
https://drive.google.com/open?id=14CKkA9wONSfxuFeMkWFol67pNa30JSeT

<br>
#### Steps to Reproduce

1. precompute ersp measures
2. set erspparams:  pop_erspparams(STUDY,'subbaseline','on');
3. plot ersp data with std_erspplot

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*R2018a*] |
| EEGLAB version | [*v2019.0*] |

arnodelorme, 2019-12-10:
Hi Yuri,

I have tried with our tutorial study and it works fine. Would you mind to share your study. I know it can be big, but we cannot reproduce the problem without your study.

Arno

yurimaxatf, 2019-12-11:
I apologize. It seems that I hastened to open this issue. The problem was in my study files. 
Thanks for answering so quickly.

closed 2019-12-11 by yurimaxatf