sccn/eeglab #103: Error loading .edf files

javadi, 2019-12-23:
Hello

I have a set of 33 .edf files. I can load all the data except three of them. I can load these files in SPM with no problems. The error that I receive is:

`EEGLAB error in function edfannot2evt() at line 57`

I also receive the following warning in the Command Window

```
Reading data file header...
Warning: EDFANNOT2EVT.M is currently not supported! If there are problems reading EDF files, 
Warning: download and install mexSLOAD.mex* from https://pub.ist.ac.at/~schloegl/src/mexbiosig/ 
```

Here I've attached one sample file. I downloaded EEGLAB 19-1 just today. Thanks. 

Greetings
Amir-Homayoun

[e31.zip](https://github.com/sccn/eeglab/files/3995465/e31.zip)

arnodelorme, 2020-01-07:
I was able to import the file using the BDFimport plugin (which also imports EDF). It is a strange looking file with lots of gaps. Also copying alois @schloegl

arnodelorme, 2020-01-07:
We will not fix BIOSIG, so I am closing this report and Alois might be able to look into it.

closed 2020-01-07 by arnodelorme