sccn/eeglab #106: EEGLAB error in function contour() at line 46

SergeyA1973, 2019-12-29:
### Description
This error is appearing at the plotting of the spectral graphs. In the window write Undefined variable "matlab" or class "matlab.gaphics.chart.internal.contourobjHelper.
I think, that is a bug, because it's appearing only in Linux version (9.5)
How it possible to resolve it?

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

SergeyA1973, 2019-12-29:
Linux MATE 18.04
Matlab 9.5

nucleuscub, 2020-01-07:
@SergeyA1973, 
 This may not be an EEGLAB issue. The function contour() does not belong to EEGLAB, but to MATLAB core functions instead.  This may be an issue with the paths associated with the MATLAB installation. There are a few posts on the web to address similar issues (e.g., [here).](https://www.mathworks.com/matlabcentral/answers/434401-undefined-variable-matlab).
 Hope this helps,
R

closed 2020-01-07 by nucleuscub