sccn/eeglab #107: Show trial number at bottom of `eegplot()`

mattansb, 2020-01-06:
### Description

Currently, when looking at many trials at once, with many events, it often impossible to see the trial number - which is hidden by the event flags:

[image]

#### Expected behavior:

This could be solved by having the trial numbers on the bottom of the plot, where they would not be obscured by anything.

#### Actual behavior:

(Trial number is obscured)

#### Versions

|   |   |
|----| ---- | 
| OS version     | Win10 |
| Matlab version | R2017b |
| EEGLAB version | 2019.1 |

nucleuscub, 2020-01-07:
@mattansb, 
 The lower x-axis is for displaying the latencies. You may shrink the time displayed by adjusting the **Time range to display** in the menu **Settings**, that way you will see clearly all the labels.
 Hope this helps,
R

closed 2020-01-07 by nucleuscub

mattansb, 2020-01-07:
> You may shrink the time displayed by adjusting the Time range to display 

True - but would like to look at many trials at once, without having to switch between display settings to "zoom in" like that.
Other solutions:
- Perhaps the zoom in feature could also scale the events and would allow with simple click-and-drag to see the trial number.
- like the latency / amp / channel values in the bottom-right, showing the values at current mouse location could also display the current trial number over which the mouse is.