sccn/eeglab #108: `eegplot()` extremly slow in 2019.1

mattansb, 2020-01-06:
### Description

Since switching to V2019.1, the data scrolling via `eegplot` has become extremely slow - whereas in previous versions plotting was more-or-less instantaneous when plotting the next couple of trials, now takes many seconds to plot (12 seconds to plot 20 trials) - which is honestly unbearable.

#### Steps to Reproduce

We've tried on 3 computers, with different sized data sets - in all the same slowness in experienced.

#### Expected behavior:

Ability to scroll epoched data fast.

#### Actual behavior:

Data scrolling is painstakingly slow.

#### Versions

|   |   |
|----| ---- | 
| OS version     | Win10 |
| Matlab version | R2017b |
| EEGLAB version | 2019.1 |

nucleuscub, 2020-01-07:
Hi @mattansb,
I'll try with Win10, but so far with MACOS there is no issue at all. There has not been any change in eegplot.m between the 2019.0 and 2019.1 to justify the slowdown.
Question here:
Are the three computers you tried running on windows?
What was the previous EEGLAB version installed and running without issues?
Thks,
R

nucleuscub, 2020-01-07:
One more... how many events do you have in the dataset?

arnodelorme, 2020-01-07:
Yes, the last change to eegplot.m is dated of April 2019 which was before 2019.0 was released. If you can run the profiler, we can assess what is slowing things down. My guess is a problem between Matlab and Windows display drivers. Maybe change the default renderer for figures.

set(0, 'defaultFigureRenderer', 'painters')

or

set(0, 'defaultFigureRenderer', 'zbuffer')

Arno

mattansb, 2020-01-08:
@nucleuscub and @arnodelorme 

It seems the problem is different than I thought: **The *later* the epoch, the *longer* it takes to plot it.** That is, clicking `>>` from 1 to 6 plots faster (less than 2 seconds) than from 701 to 706 (~20 seconds).
[I've changed the name of the issue accordingly]

- I've tried this on eeglab 14.1, and the problem was there too. We just didn't notice it because previous projects had a smaller number of epochs.
- All three computers we tried are running on windows.
- Trying `set(0, 'defaultFigureRenderer', 'painters')` / `set(0, 'defaultFigureRenderer', 'zbuffer')` had no effect.

nucleuscub, 2020-01-08:
Thanks, @mattansb , 
 Can you please make a profile of the scrolling on eegplot.
 From the matlab command line activate the profile (profile on), then launch eegplot and start scrolling as to reproduce the issue. When done,  use 'profile review' to generate the report. Please share that with us. I'm afraid the fuinction axes may be taking too long for some of the graphical proccesing in Windows. In any case, let's see the profile first.
 Thanks,
Ramon

mattansb, 2020-01-09:
@nucleuscub thanks!
Here is a zip of the following:

```Matlab
profile on
eeglab
% open data scroll and scroll a couple of times
% jump to trial 300 and scroll a couple of times
% jump to trial 700 and scroll a couple of times
profile viewer
profsave
```

[profile_results.zip](https://github.com/sccn/eeglab/files/4039445/profile_results.zip)

arnodelorme, 2020-01-10:
This is because you have a lot of events. Most of the time the function is evaluating involve plotting events. I am not sure about the proportional time with how far it is in the data. This does nor make sense to me.

Would you mind to try the EEGbrowser plugin and tell us if has the same problem?

mattansb, 2020-01-11:
`EEGbrowser` browser does not have the same problem, but it is not suitable for my needs as I need to be able to see the trial number (#107 ) which `EEGbrowser` does not show (only seconds).

arnodelorme, 2020-01-11:
Would you mind to share a dataset for testing purposes.

Arno

mattansb, 2020-01-13:
Sure:
- [601_fil_seg.set](https://drive.google.com/a/post.bgu.ac.il/file/d/1kB6ORp3pe2k6gwHbZOV1iIHKpvhuxo6k/view?usp=sharing)
- [601_fil_seg.fdt](https://drive.google.com/a/post.bgu.ac.il/file/d/1kAiESgpictVDB83XCBAy8rHH3Wx2wxhW/view?usp=sharing)

arnodelorme, 2020-01-13:
I have corrected a bug in c600a5376 associated with scanning all events.

Arno

mattansb, 2020-01-14:
@arnodelorme Thank you so much! Works great!

closed 2020-01-14 by mattansb