sccn/eeglab #109: EEGLAB ERROR

Mehrdadndr, 2020-01-07:
Dear Sir/Madam
I have an issue to import some data not all. I received error message that I attached. What would be the problem? 
Best,
Mehrdad
[image]

nucleuscub, 2020-01-07:
@Mehrdadndr ,
I think the issue may be with the BIOSIG toolbox. But this is me just guessing. There are basic guidelines to follow when submitting a bug report. Enough information should be provided to ensure we can reproduce the issue, asses it and solve it (e.g. sample data, code). These guidelines are in the form when submitting a report. Please review that so we can help you in troubleshooting the issue.
 Thanks,
R

arnodelorme, 2020-01-07:
Yes, your error message indicate to install mexSLOAD.mex. You might need to install this tool and import the data from the command line. You must be trying to import a rare data format.

closed 2020-01-07 by arnodelorme

Mehrdadndr, 2020-04-04:
Dear Arnaud,

I analyzed my data based on the image you received. I would like to know why the topographic maps are not completely the same while the range of the scale is the same in three comparisons? Could you please explain it to me?

Best wishes,
Mehrdad Naderi
[cid:f78cb3f3-e7ec-4a90-aad1-5b4b86fd875f]