sccn/eeglab #111: Validation in checkset function ruins chanlocs structure on XDF import from gtec's Nautilus

shirareznik, 2020-01-11:
### Description

When importing LSL XDF file from  gtec's Nautilus recordings with 38 electrodes (32 +6 additional data channels)- the imported eeg structure doesn't contain chanlocs, due to eeglab's checkset function removing it!
(I can provide a fix! Done the analysis already and fixed for myself) 

<br>
#### Steps to Reproduce

1. Recorded data using gtec's Nautilus using LSL module, including additional channels (38 channels overall)
2. imported the data using pop_loadxdf

<br>
#### Expected behavior:
import results in a eeg structure WITH chanlocs field

<br>
#### Actual behavior:
import results in a eeg structure with NO chanlocs field

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     |Windows 7|
| Matlab version | 2017|
| EEGLAB version | 14.1.2|

(might add an example file tomorrow)

Thanks

arnodelorme, 2020-01-13:
This is because the channel location is not compatible with the number of data channels. Some channels are probably fiducials but are not recognized as such, so EEGLAB thinks the two information are not compatible. Would you mind to share a data file?

shirareznik, 2020-01-14:
Yes- will send privately (please erase the file following the analysis)
Thanks

arnodelorme, 2020-01-19:
I have fixed the problem in

https://github.com/arnodelorme/xdf-EEGLAB/blob/master/eeg_load_xdf.m

We will release an updated version of the plugin.
Thank you,

Arno

arnodelorme, 2020-01-19:
Actually you can now update to version 1.15. I have deleted your file. Let us know,

Arno

closed 2020-01-19 by arnodelorme