sccn/eeglab #112: How can i fix this error? I can't fix this error.

bambarBrl, 2020-01-13:
help me!!!!!!!

EEGlab error in function subsasgn() at line 66:

Matrix index is out range for deletion.

if you think this is a bug, please submit a detailed description of how to reproduce the problem(and upload a small dataset) at https://github.com/sccn/eeglab/issues

arnodelorme, 2020-01-13:
We need more context. What did you do to generate the error? Make sure you have the following tow options unset (in EEGLAB menu "File > Preferences").

"If set, use memory mapped array under Matlab 7.x. This may slow down some computation (beta)."
"If set, use the EEGLAB EEG object instead of the standard EEG structure (beta)."

Best wishes,

Arno

arnodelorme, 2020-01-19:
Please reopen the issue if you are able to provide more details.

closed 2020-01-19 by arnodelorme

bambarBrl, 2020-01-21:
[image]