sccn/eeglab #115: distance (line 93) error

jbasso82, 2020-01-16:
I am working with a continuous EEG dataset. I am simply trying to use the "Reject data using Clean Rawdata and ASR" tool, but I am getting the following error: "distance (line 93) error".

Steps to reproduce:
1. Tools
2. Reject data using Clean Rawdata and ASR
Note: All default settings were kept

I expected to have a clean dataset that was ready to be processed through ICA; however, I got the following error:

distance (line 93) error
Not enough input arguments
(Error occurred in function distance() at line 93)

Using Version 19_1 on a Mac.

Link to data: https://drive.google.com/drive/u/1/folders/14AJurQ-ITaX_ZKOfUOTMZaPmTPDL0drQ

nucleuscub, 2020-01-18:
@jbasso82 , Which version of the plug-in Clean Raw data are you using? If it is not 2.1, please update and try again. Let us know how it goes after that.
 Thanks,
Ramon

jbasso82, 2020-01-19:
I am using version 19_1 of eeglab. This is the most updated version. I am simply accessing the "Reject data using clean raw data and ASR" under tools. Should I be doing something differently?

arnodelorme, 2020-01-19:
The data is not accessible.

jbasso82, 2020-01-20:
Hi,

I just made the folder public. Thank you!

arnodelorme, 2020-02-19:
I have downloaded your datasets and run some tests - but there no error with ASR. I have also checked the current clean_rawdata plugin being released. There used to be an issue with clean_rawdata "distance" and newer versions of Matlab but this was fixed in version 2.1. Are you using the latest version of clean_rawdata?

Closing for now, but feel free to reopen.

closed 2020-02-19 by arnodelorme