sccn/eeglab #116: EEGLAB function has same name as a MATLAB builtin

paulapbrooks, 2020-01-17:
### Description

Hello, I just downloaded the latest version of EEGLAB (eeglab2019_1) to run on MATLAB R2016b. The installation went well and I was able to open the toolbox. However, whenever I open MATLAB for the first time, I keep getting the following warning:

[image]

I initially just thought about just ignoring this warning. However, whenever I use **addpath()** in my code, I get the following warning:

[image]

In my function **init_exp**, this is line 127: 
[image]

_What can I do to fix this warning?? Thanks!_

#### Steps to Reproduce

1. Install EEGLAB on MATLAB 2016b
2. Restart MATLAB -- warning should appear on the command window
2. Try to temporarily add another path to MATLAB -- warning should appear again

#### Expected behavior:

I didn't expect to see a warning about an EEGLAB function conflicting with MATLAB functions...

#### Actual behavior:

But I received a warning...

#### Versions

|   |   |
|----| ---- | 
| OS version     | macOS 10.12.6 |
| Matlab version | 2016b |
| EEGLAB version | 2019 |

nucleuscub, 2020-01-18:
@paulapbrooks, it is probable that you installed EEGLAB by adding all the subfolders to the path, which should not be done. This may have added ICLabel functions to the path and ended up with this conflict. You may want to re-installl EEGLAB by adding just the EEGLAB folder to he path. 
Best,
Ramon
@lucapton, these functions belong to the plug-in ICLabel, can I go ahead and rename it?

paulapbrooks, 2020-01-18:
@nucleuscub WOW! Thank you so much for your help, Ramon. I did what you suggested and it all works as it should. I am no longer getting the warning :)

Take care,
Paula

closed 2020-01-18 by paulapbrooks

lucapton, 2020-01-18:
@nucleuscub Go ahead and change the names as you like. I don’t think people would be using those variables directly so it shouldn’t cause a problem. Are you going to push a new version to the plugin manager too?