sccn/eeglab #117: help me

bambarBrl, 2020-01-21:
if you think this is a bug, please submit a detailed description of how to reproduce the problem(and upload a small dataset) at https://github.com/sccn/eeglab/issues

bambarBrl, 2020-01-21:
[image]

nucleuscub, 2020-01-21:
@bambarBrl , 
We need more information on the context of this issue to be able to help you (please read and follow the form when submitting an issue). The most probable thing, though,  is a mismatch of the events and data latencies. But without the dataset, it is impossible to confirm.  Please consider sharing the dataset with us, so we can look into it.
 Thanks,
Ramon

arnodelorme, 2020-01-24:
In the EEGLAB Preferences make sure the option to use memory mapping and EEG objects (option number 5 and 6) are unchecked. Let us know,

Arno

arnodelorme, 2020-02-19:
Closing bug because no response. Feel free to reopen.

closed 2020-02-19 by arnodelorme