sccn/eeglab #128: Data range mismatches with actual recording duration

Yifangx1, 2020-01-29:
Hello,

I'm working with an edf file exported by EmotivPro. The duration of the recording is 4 minutes and 55 seconds. When I imported the edf file to EEGLAB, the data range appeared to be [0 286]. However the full data range should be [0 295]. Could anyone tell me why the data range showed in EEGLAB is shorter than actual recording duration? I checked the exported excel file and the data range in the excel file seemed correct.

Any help is appreciated. Thank you in advance.

arnodelorme, 2020-02-19:
Would you mind to provide a data file? Also, which importer are you using in EEGLAB (there are 2 of them - BDFplugin and BIOSIG and both of them import EDF). For seeing the extended length, what does the following EDF viewer program return?

https://bilalzonjy.github.io/EDFViewer/EDFViewer.html
https://www.teuniz.net/edfbrowser/

Yifangx1, 2020-02-24:
Hello Arnaud,

Thank you for your reply. I loaded the data into EDFViewer and the duration of the recording is indeed 00:04:46. I assume that there could be some problems when recording the signal. It should not the problem of EEGLAB.

arnodelorme, 2020-02-25:
OK, thank you for letting us know. I am closing this report now.

closed 2020-02-25 by arnodelorme