sccn/eeglab #129: Spectra comp issue

NathiMagubane, 2020-02-05:
New to EEG Lab and Matlab in general, so I do not know how to explain the issue all too well. 
I have been trying to spectra map my dataset, but I am presented with the following error: 

EEGLAB error in function spectcomp() at line 941: 

Undefined function 'pwelch' for input arguments of type 'double' 

| OS version     | MacOS Mojave
| Matlab version | R2019b
| EEGLAB version | v2019.1

zegamer, 2020-02-13:
Try to install Signal Processing Toolbox from the Add-on manager (Home > Add-on > Get Add-ons > Search for SPT > Install). It worked for me afterwards.

arnodelorme, 2020-02-19:
I agree. We will investigate.

callen07, 2020-05-22:
Hi.  I have the same problem with the function findpeaks:

EEGLAB error in function findpeaks() at line 231:

Undefined function 'findpeaks' for input arguments of type 'double'

I have the SPT from Matlab, I asked them, because seems that the toolbox is not there, and I payed for that.  

>> which findpeaks -all  Running it before and after open eeglab we realized that EEGlab seems to remove some folders from your search path, because after open EEglab does not find any functions

This also affect another part of the analysis that I am pretending to do, so if you can help me to solve this, would be amazing.

arnodelorme, 2020-05-22:
I have fixed the issue with pwelch. I have also added additional warnings.

closed 2020-05-22 by arnodelorme

callen07, 2020-05-23:
What additional warnings?  Where I can find them, to solve my problem?