sccn/eeglab #131: EEGLAB and NetStation 5 Compatibility Issues

rpriefer, 2020-02-14:
### Description

When processing raw .mff data in EEGLab and then importing it into NetStation 5 for segmentation and baseline correction, NetStation 5 will sometimes crash on seemingly random files, as all of them are being processed the exact same way with the same parameters. Sometimes, it won't crash, but segmentation will spit back an error. Using the NetStation 5 File Validator, we receive a peculiar "error" that seemingly shouldn't cause issues, but it does.

See link below for some images.
https://drive.google.com/drive/folders/1icmLmz04ApN4_4C2_36KepGXkKNE0lc_?usp=sharing

<br>
#### Steps to Reproduce
These are the processing steps we have been using:

1. Mark some bad channels on the raw .mff file in NS5
2. Import EGI/Philips .mff file into EEGLAB. Select "code" and "label" to import as well.
3. Filter .3 - 30 Hz
3. Reject continuous data by eye
4. ASR (without bad channel removal; otherwise default settings)
5. Rereference
6. ICA decomposition
7. ICLabel and remove components
8. Export .mff file
9. Import .mff file into NetStation 5 for final processing steps
10. Segment data**** (where the issue arises, never had issues with the steps preceding or succeeding this part)
11. Baseline correction
12. Averaging
13. Text file export for plotting in R Studio

<br>
#### Expected behavior:
I expect the file to go through the segmenting steps without any issue.

<br>
#### Actual behavior:
Two things may happen, which seems to be random:
1. An error in segmenting occurs and we receive the following message: "Info provided: EGISegmentationApplication: error: No segments were created" 
When putting this file into the file validator, we receive this message: "Problem: Epochs based on signal bin versioning does not match the data in the Epochs.xml file" "Solution: This is for your information only. It should not matter in most cases"

OR

2. When attempting to segment, Netstation 5 Tools crashes and provides the option to ignore, reopen, or send an error report (which has been done already).

These issues arise only on some of our files, and not others, despite all of them being obtained on the same acquisition system (although this occurs randomly for our new acquisition system as well), and processed the same exact way. Sometimes, we have reprocessed the files again from the beginning, and they will work. There does not seem to be a rhyme or reason except perhaps that it is being mostly processed in EEGLAB and partially in NS5.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | macOS High Sierra 10.13.4 |
| Matlab version | R2018b|
| EEGLAB version | 2019.1 |

arnodelorme, 2020-02-19:
Moved to the MFFmatlabio plugin

https://github.com/arnodelorme/mffmatlabio/issues/19

Would you mind to provide a file that causes problem for testing purposes. You can drag and drop below if necessary 

https://drive.google.com/open?id=1mrHVar2lBju7qM9uqqjJxpYPVuWStP-5

PLease do not comment on this bug report but on this one intead https://github.com/arnodelorme/mffmatlabio/issues/19

closed 2020-02-19 by arnodelorme

rpriefer, 2020-02-19:
Priefer_AV RAW DATA.zip
<https://drive.google.com/a/adelphi.edu/file/d/1qne7yhz6U0BU6GCqn4Abvn35kRNIPEBP/view?usp=drive_web>