sccn/eeglab #135: .csv and edf extension files did not import

Rneem, 2020-02-28:
### Description:
 I tried to import two kind of eeg files .csv and .edf
but show me a problem
for .edf the message is (BIOSIG extention is not onatalled, do you want to download it now?) i already extract the zip file in the plug-in file but i don/t know the problem

also for .csv file i alreday plug-in musemonitor3.1 it show me to contact this page 

please find the error i captured in the attachment file
best regards

[image]
[image]
[image]

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

Rneem, 2020-02-28:
matlav version 2019a
EEGLAB version eeglab2019_1
OS windows 10

arnodelorme, 2020-03-02:
Musemonitor plugin will only import .csv file generated by the Musemonitor program. To import any CVS file, import it on the matlab command line using

data = csvread(...);

Then use the EEGLAB interface to import the array using Menu item File > Import data > Using EEGLAB ... > From ASCII/float file or Matlab array.

For the problem with EDF, something prevents EEGLAB from accessing the internet and download the necessary plugin for importing EDF file. It is likely your firewall. You can download the zip file manually at https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php, uncompress it in the EEGLAB plugins folder and restart EEGLAB. 

Best wishes,

Arno

closed 2020-03-02 by arnodelorme