sccn/eeglab #137: Study fails with Matlab 2019a, but runs with 2019b

JohnAtl, 2020-03-03:
### Description
When running plot channel measures on 2019a, there is an assignment error. The same study runs without error on 2019b.

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

[image]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [macOS 10.15.3] |
| Matlab version | [2019a and 2019b] |
| EEGLAB version | [2019.0] |

arnodelorme, 2020-03-06:
That is surprising. Share the study if you can.

JohnAtl, 2020-03-07:
[vfnf.study.zip](https://github.com/sccn/eeglab/files/4302586/vfnf.study.zip)

JohnAtl, 2020-03-07:
I'll email it to you.

arnodelorme, 2020-04-16:
Hi John,

Did you send me the study so I could have a look?

Arno

JohnAtl, 2020-04-16:
Hi Arno,

I thought I had deleted those files and lost them for good, but found a 
backup on Backblaze.
I've uploaded them to the shared folder you sent.
Thanks!

https://drive.google.com/drive/folders/1mBzosm2q1-M2pWpmbfnzxIeN3IVbT0LY

Regards,
John

arnodelorme, 2022-02-24:
Closing this bug report

closed 2022-02-24 by arnodelorme