sccn/eeglab #138: std_preclust error: subscripted assignment dimension mismatch

Naama2020, 2020-03-05:
### Description
https://sccn.ucsd.edu/bugzilla/show_bug.cgi?id=1770
[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
I have 
[Dipole_8-13HZ.zip](https://github.com/sccn/eeglab/files/4293289/Dipole_8-13HZ.zip)
[image]

a similar issue with preclustering of my data (31 datasets). Attached is a screenshot of the error message and the compressed dataset.
I should note that I keep getting a bug where in STUDY.cluster.sets I get NAN in the same sets, and I manually fix this matrix. Previously, using this solution has enabled preclustering and clustering, but with dipole data this error message appears.
[Dipole_8-13HZ.zip](https://github.com/sccn/eeglab/files/4293268/Dipole_8-13HZ.zip)

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2020-03-06:
Everything looks ok for me in the file you have shared. Unfortunately, we would need to see the full study to really assess the problem. Any chance you could share it. We can provide a place to upload it if it is large.

Best - Arno

Naama2020, 2020-03-06:
I can upload everything, but it is a lot.

Naama

arnodelorme, 2020-03-06:
Can you email us at eeglab@sccn.ucsd.edu
I do not want to post the location for uploading data on Github

Arno

arnodelorme, 2020-03-06:
Hi Naama,

I have looked at the files, but the STUDY file points to other files.

1. I have put all your files in the same folder
2. I have imported them all in EEGLAB (you can import multiple files if they are in the same folder)
3. I created a study with the loaded files
4. I precomputed ERP for components
5. I preclustered based on ERPs
6. I clustered

No problem on my side.
This was with EEGLAB 2019.1
Best wishes,

Arno

Naama2020, 2020-03-06:
Do you think I could be having problems because I use EEGLAB version
14.1.2? Should I try to download the latest version of EEGLAB?

‫בתאריך יום ו׳, 6 במרץ 2020 ב-18:32 מאת ‪Arnaud Delorme‬‏ <‪
notifications@github.com‬‏>:‬

arnodelorme, 2020-03-06:
Yes, it is possible. You should definitely use the latest version.
Make sure you recreate the study. Your study file points to the wrong files.
Best wishes,

Arno

Naama2020, 2020-03-06:
Hi,

I downloaded the latest version of eeglab and organized the files in the
same folder, as instructed.
All went well, but when I pre-computed component measures (ERP, spectra,
ERSP, ITC, and scalp mas) this is the error message I received:

"std_precomp (line 290) error

ICA decompositions must be identical if several datasets are concatenated
for a given subject."

I would send this as a separate eeglablist error, but it might be linked to
the similar issue I sent (since it occurs when I pre-compute).

Thank you for your assistance,
Naama

‫בתאריך יום ו׳, 6 במרץ 2020 ב-19:01 מאת ‪Arnaud Delorme‬‏ <‪
notifications@github.com‬‏>:‬

arnodelorme, 2020-03-06:
Would you mind to try with the zipped version at

https://github.com/sccn/eeglab

Select clone or download and download a zip file. This is the same version I am using.
Also I noticed that in the STUDY you sent there was a pilot subject but this was not the case in the files you shared.
Best wishes,

Arno

Naama2020, 2020-03-06:
I downloaded the zip file of the latest version from the link you provided,
as you instructed - I still get the same error message.
The pilot files are actually the first subject, and the numbers of subjects
in the file names do not match the subject numbers in the study, because
some of the participants in the study were discarded. I have a document
that indicates which participant numbers go with each of the subject
numbers in the study in eeglab. But thank you for noticing.

Naama

‫בתאריך יום ו׳, 6 במרץ 2020 ב-21:07 מאת ‪Arnaud Delorme‬‏ <‪
notifications@github.com‬‏>:‬

arnodelorme, 2020-03-06:
Can you provide the exact steps you followed. Did you follow the steps below?

> 1. I have put all your files in the same folder
> 2. I have imported them all in EEGLAB (you can import multiple fles if they are in the same folder)
> 3. I created a study with the loaded files
> 4. I precomputed ERP for components
> 5. I preclustered based on ERPs
> 6. I clustered

Thank you,

Arno

Naama2020, 2020-03-06:
So I followed the exact steps instructed until step 4. I precomputed ERP
but also spectra, ERSP, ITC, and scalp maps. I got this error:
"std_precomp (line 290) error

ICA decompositions must be identical if several datasets are concatenated
for a given subject."

I then tried precomputing for ERP only. I got the same error.

I should note that I have had a problem in the past with NANs appearing in
STUDY.cluster.sets. They appear in the same values every time. These NANs
appear in this study too. Previously, I solved this problem by moving all
the lines with values into the right order and erasing all extra columns.
This would solve the problem. However, since I have added the dipole info
to the datasets it doesn't seem to help.

Naama

‫בתאריך שבת, 7 במרץ 2020 ב-0:05 מאת ‪Arnaud Delorme‬‏ <‪
notifications@github.com‬‏>:‬

arnodelorme, 2020-03-07:
I was able to compute ERSP.
I have NaN as well in STUDY.cluster(1).sets but it works.

You might want to fix the NaNs first. For this all datasets of a given subject must have the same ICA decomposition.

Best wishes,

Arno

Naama2020, 2020-03-16:
Hi,

I hope it is okay I ask about this topic again. I tried to pre-compute
component measures again with only ERP, Spectra, and Scalp maps and
received the attached error message.
In light of this error, I attempted to run a smaller study with a smaller
number of subjects, just to see if having an identical ICA decomposition
would indeed enable me to pre-compute the measures. However, I seem to have
a problem: after running ICA decomposition for all 4 datasets of each
subject and then using DIPFIT for each dataset - the 'identical ICA' field
changes from 'yes' to 'no'. Does this mean that if I want to run my study
with an idential ICA decomposition per subject I have to give up the DIPFIT?

Thank you for your assistance,
Naama
‫בתאריך שבת, 7 במרץ 2020 ב-3:49 מאת ‪Arnaud Delorme‬‏ <‪
notifications@github.com‬‏>:‬

arnodelorme, 2022-04-19:
Closing this report

closed 2022-04-19 by arnodelorme