sccn/eeglab #139: pop_runica error line 538

annmacfarlane, 2020-03-05:
### Description
as a part of my preprocessing for my EEG data in EEGLAB i have been using run ICA for the 2019.0 version and decompose data by ICA for the 2019.1 version. 
recently i have encountered a pop up error:

"EEGLAB error in function popo_runica at line 538
index in position 1 exceeds array bounds (must not exceed 1)."

In MATLAB it gives me this:

"Attempting to convert data matrix to double precision for more accurate ICA results.
Data rank (12) is smaller than the number of channels (129).
runica(): unknown flag>> "

My datasets have 129 channels and their structural properties in my workspace are exactly the same as that of the previous datasets which this function has worked on effectively so am hoping somone could explain why this error has seemingly appeared from no where?

I am a MATLAB beginner so it may well be an easy fix i havent come across.
| Matlab version | [R2019a] |
| EEGLAB version | [2019.1] |

any help is greatly appreciated!

Ann

arnodelorme, 2020-03-06:
Dear Ann,

Could we have access to the problematic dataset for testing purposes. You can zip it and drag it here if it is not too large.

Cheers,

Arno

annmacfarlane, 2020-03-06:
Dear Arno,

Thanks for the prompt reply. 
Here is the dataset
[EEG_dataset.zip](https://github.com/sccn/eeglab/files/4297444/EEG_dataset.zip)

Thanks,
Ann

arnodelorme, 2020-03-06:
Would you mind to send the associated .fdt file as well?

Arno

annmacfarlane, 2020-03-09:
Hi Arno,

I'm having trouble doing this due to the dataset size, even in a zip folder its still 59,007KB so it won't let me upload it or email it to you. Should i reduce the dataset size by cutting it in time?

Best,
Annabel

annmacfarlane, 2020-03-10:
Hi Arno,

Here is a 60 second extract of the data, i am still unable to run ICA on this due to the same errors despite its much smaller size.
[EEG_dataset.zip](https://github.com/sccn/eeglab/files/4311528/EEG_dataset.zip)

Thanks so much for your help,
Ann

arnodelorme, 2020-04-01:
Hi Ann,

I have imported your dataset in EEGLAB 2019.1 (the version you can download from GITHUB when you download the zip file) and was able to run ICA no problem. I have tried on Mac and Windows, with all the default values. In both case, the function suggested that I reduce the rank to 128 which I accepted.

Let me know how we can help further.
Best wishes,

Arno

closed 2020-04-01 by arnodelorme